Avnu/OpenAvnu #831: Code compliance

2raghu, 2018-09-18:
Hi,
Is this source code fully compliant to 802.1AS spec? Are there any companies who are using this code for their product and is in field? I'm planning to port the code to FreeRTOS for one my development board(samv71 xplained).

Pls clarify.
Thanks

andrew-elder, 2018-09-18:
Yes, the 802.1AS implementation is fully compliant with the 802.1AS spec. And, yes, there are companies that have this code running in real products in the field.

2raghu, 2018-09-19:
Thank you for the clarification.
What about 802.1ASrev, is it compliant?
Thanks for your time.

christopher-s-hall, 2018-09-19:
No. 802.1AS-rev's "major" feature addition is support for multiple domains. I think to support multiple domains using gPTP an additional arbitration layer would need to be added between the packet socket and the port object.

2raghu, 2018-09-20:
Thanks a lot.
Sorry for the trouble,One last query, Does this code support only for endpoints or can be used for switches too?
Thanks again.