Avnu/OpenAvnu #848: Ubuntu 18.04, I218 simple talker not working

shirArg, 2018-11-22:
Hi, 
I installed everything, and trying to use this project, but things are not working for me. I'm working on Ubuntu 18.04, with Intel I218-LM. 

When trying to use startup.sh I get:
`shir@shir-Latitude-E5450:~/git/OpenAvnu/lib/igb_avb/kmod$ sudo ./startup.sh eno1
rmmod: ERROR: Module igb is not currently loaded
rmmod: ERROR: Module igb_avb is not currently loaded
insmod: ERROR: could not load module ./igb_avb.ko: No such file or directory
./startup.sh: line 20: /sys/class/net/eno1/queues/tx-1/xps_cpus: No such file or directory
./startup.sh: line 21: /sys/class/net/eno1/queues/tx-2/xps_cpus: No such file or directory
./startup.sh: line 22: /sys/class/net/eno1/queues/tx-3/xps_cpus: No such file or directory`

When trying to `make kmod` (I don't know if I should):
`shir@shir-Latitude-E5450:~/git/OpenAvnu/lib/igb_avb/kmod$ make kmod
cc1: error: code model kernel does not support PIC mode
/bin/sh: 1: [: -ge: unexpected operator
Makefile:202: *** *** Aborting the build. *** This driver '' is not supported on kernel versions older than 2.4.21, because this driver requires NAPI support..  Stop.
`
I tried to solve it using the patch described[ here](https://askubuntu.com/questions/851433/kernel-doesnt-support-pic-mode-for-compiling), but it didn't work. 

I cloned gptp repository too. After a while I copied it into the daemons folder in the OpenAvnu folder, I thought it might solve some problems. When trying to use daemon_cl, I get-
`shir@shir-Latitude-E5450:~/git/OpenAvnu/daemons/gptp/linux/build/obj$ sudo ./daemon_cl eno1
INFO     : GPTP [17:27:25:354] gPTP starting
ERROR    : GPTP [17:27:25:354] Using clock device: /dev/ptp0
STATUS   : GPTP [17:27:25:354] Starting PDelay
ERROR    : GPTP [17:27:25:450] Error (TX) timestamping PDelay request, error=-72	
ERROR    : GPTP [17:27:26:517] Error (TX) timestamping PDelay request, error=-72	
ERROR    : GPTP [17:27:27:584] Error (TX) timestamping PDelay request, error=-72	
STATUS   : GPTP [17:27:28:354] *** Announce Timeout Expired - Becoming Master
STATUS   : GPTP [17:27:28:354] New Grandmaster "20:47:47:FF:FE:E3:0D:AF" (previous "00:00:00:00:00:00:00:00")
EXCEPTION: GPTP [17:27:28:450] PDelay Response Receipt Timeout
ERROR    : GPTP [17:27:28:650] Error (TX) timestamping PDelay request, error=-72	
EXCEPTION: GPTP [17:27:29:517] PDelay Response Receipt Timeout
ERROR    : GPTP [17:27:29:716] Error (TX) timestamping PDelay request, error=-72	
EXCEPTION: GPTP [17:27:30:584] PDelay Response Receipt Timeout
ERROR    : GPTP [17:27:30:783] Error (TX) timestamping PDelay request, error=-72	
EXCEPTION: GPTP [17:27:31:650] PDelay Response Receipt Timeout
ERROR    : GPTP [17:27:31:849] Error (TX) timestamping PDelay request, error=-72	
EXCEPTION: GPTP [17:27:32:717] PDelay Response Receipt Timeout
ERROR    : GPTP [17:27:32:915] Error (TX) timestamping PDelay request, error=-72	
EXCEPTION: GPTP [17:27:33:783] PDelay Response Receipt Timeout
ERROR    : GPTP [17:27:33:981] Error (TX) timestamping PDelay request, error=-72	
EXCEPTION: GPTP [17:27:34:849] PDelay Response Receipt Timeout
ERROR    : GPTP [17:27:35:047] Error (TX) timestamping PDelay request, error=-72	
^CERROR    : GPTP [17:27:35:604] Exiting on 2`
 I found that issue #265  is related to this, but I couldn't find the solution there. 

When trying to run the simple_talker, I get-

`shir@shir-Latitude-E5450:~/git/OpenAvnu$ sudo ./run_simple_talker.sh eno1
connect failed (Success) - are you running as root?`

I don't know if I have one problem, or does this all part of the same one, but I couldn't figure out any of it. 

Any ideas what can I do?...

Thanks

pinealservo, 2018-11-26:
Intel i218 is not the same as Intel i210.  The igb kmod only works (as far as I'm aware right now) on i210. The simple talker and simple listener examples require the igb module, so those won't work at all.

You may be able to get the avtp_pipeline example working if you can get gPTP to work (and the i218 should support PTP) but it's not really production-ready without a lot of tweaking to ensure that you're meeting latency targets consistently.

I have no idea why your gPTP daemon is not working.  You aren't running in a virtual machine, are you? Somehow the PTP support for the NIC in the kernel is not working correctly.

shirArg, 2018-11-28:
I'm not using a VM for this. 

Thank you for your response. 

Regarding the I210- how do you all managed to have 2 computers with it? I have some laptops around, and they all have I219 or I218. Do you know a way to install those network cards on laptops, or to visualize them? I really want to use TSN from something, but buying 2 new desktops with a specific network card sounds like a bit of overkill. 

Thanks

shirArg, 2019-01-03:
In the end I bought two switches... 
This can be closed now :)

closed 2019-01-07 by pinealservo