Avnu/OpenAvnu #850: AVB audio strream input chunk size?

Ravi-Teja-konda, 2018-12-04:
Hello Everyone,

What is the AVB Audio stream input chunk size in terms of samples number? 
 
Thanks,
Ravi Teja.

andrew-elder, 2018-12-04:
Typically 6 samples per channel at 48 kHz and 125 microseconds packet time. But other configurations are supported.

Ravi-Teja-konda, 2018-12-04:
Hello andrew-elder,

Thank you for the reply i just wanted to confirm

Ravi-Teja-konda, 2018-12-05:
Hello Andrew-elder,

Thank you for the reply.
I just wanted to confirm that the talker side application (wav file talker) takes the number of channels, bit size, and bit rate from the .wav file header.
At the Listener side, the listener application (alsa_listener) takes these things from the Configuration files.

So, the bit rate and sample size depends on the wav file. right?

I my understanding true?

Regards,
Ravi Teja.

andrew-elder, 2018-12-05:
Technically the .wav file format and the AVTP packet format are completely independent. Audio from a stereo .wav file could be sent in 2 one channel AVTP streams. 16-bit samples from a .wav file can be zero padded and sent in 32 bit AAF format packets. And so on.

The listener size takes packing and channel count information from the format word sent via 1722.1 and also from the information about channels in each AVTP packet.

I would take a read of IEEE 1722 and 1722.1 if you have not already done so.

pinealservo, 2018-12-05:
If you're talking about the avtp_pipeline code, both talker and listener ends have both an "interface" component that selects where the data comes from or goes to, and a "mapper" component that defines how inputs/outputs map to some AVTP stream format. Both interfaces and mappers have their own configuration files and need to be set appropriately for the desired streaming.

The wav file interface (which I haven't used myself) does appear to read from the header, but it also has config file parameters for the audio format. I haven't got time right now to check what happens if you don't set the config file parameters or if they don't match what's in the header, but you might try setting both at first to the correct values to be safe.

How the audio data gets packetized depends entirely on the configuration of the mapper module you choose, and that will determine how samples from the wav file appear in the packets.

Ravi-Teja-konda, 2018-12-06:
Hello Everyone,

Thank you for the reply guys.
 I have one question to @pinealservo 
 if setting  configuration for mapping module and interfaces works for packetizing the AVTP packet that is fine for me.
Can you please send the path for these config files?

Regards,
Ravi Teja.

pinealservo, 2018-12-06:
I don't know the path off the top of my head; they're .ini files and if I recall correctly copies get put into the avtp_pipeline build output directory after a successful build.