Avnu/OpenAvnu #854: gPTP: Bug mempcy for 'grandmasterIdentity'

inchoel, 2018-12-20:
https://github.com/rroussel/OpenAvnu/tree/ArtAndLogic-aPTP-changes/daemons/gptp

diff --git a/daemons/gptp/common/ptp_message.cpp b/daemons/gptp/common/ptp_message.cpp
index bc5b0e0..4406383 100644
--- a/daemons/gptp/common/ptp_message.cpp
+++ b/daemons/gptp/common/ptp_message.cpp
@@ -870,7 +870,7 @@ std::shared_ptr<PTPMessageCommon> buildPTPMessage
                               PTP_ANNOUNCE_GRANDMASTER_PRIORITY1
                               (PTP_ANNOUNCE_OFFSET),
                               sizeof(annc->grandmasterPriority1));
-                       memcpy( &annc->grandmasterClockQuality,
+                       memcpy( &(annc->grandmasterClockQuality),
                                buf+
                                PTP_ANNOUNCE_GRANDMASTER_CLOCK_QUALITY
                                (PTP_ANNOUNCE_OFFSET),
@@ -885,7 +885,7 @@ std::shared_ptr<PTPMessageCommon> buildPTPMessage
                               PTP_ANNOUNCE_GRANDMASTER_PRIORITY2
                               (PTP_ANNOUNCE_OFFSET),
                               sizeof(annc->grandmasterPriority2));
-                       memcpy(&(annc->grandmasterIdentity),
+                       memcpy((annc->grandmasterIdentity),
                               buf +
                               PTP_ANNOUNCE_GRANDMASTER_IDENTITY
                               (PTP_ANNOUNCE_OFFSET),

andrew-elder, 2018-12-20:
@inchoel - good catch. Do you want to submit a pull request to the `ArtAndLogic-aPTP-changes` branch?

inchoel, 2018-12-20:
Yes, I want to
Thank you.

inchoel, 2018-12-21:
@andrew-elder , @rrousselArtLogic 
I created a pull request for this.
https://github.com/rroussel/OpenAvnu/pull/2