Avnu/OpenAvnu #855: avnu-pipeline not working- Failed to initialize SRP

shirArg, 2018-12-24:
Hi, 
I am using two linux machines with I210 nics (one on each one). The compuiters are connected using an Ethernet cable, without a switch in the middle(!). 

I was able to run simple_talker and simple_listener, and they worked fine. Now, I'm trying to use the avtp_pipeline. 
I start the daemons using `run_daemons.sh`, and it seems that it works ok (I can see the processes in ps, for example). I also started the `igb_avb` mode using `lib/igb_avb/kmod/startup.sh`. 

I compiled the avtp_pipeline without 'AVB_FEATURE_ENDPOINT=0' (meaning it equals 1). 

When I run the `run_avtp_pipeline.sh` I get: 

argu@SHIR2:~/git/OpenAvnu$ sudo ./run_avtp_pipeline.sh enp4s0
Starting AVTP Pipeline on enp4s0
~/git/OpenAvnu ~/git/OpenAvnu
./run_avtp_pipeline.sh: line 20: .**/openavb_avdecc: No such file or directory
~/git/OpenAvnu**
argu@SHIR2:~/git/OpenAvnu$ [1545662387:160489118 OPENAVB osalTime] INFO: local_time = 1545662384217555805
[1545662387:160498483 OPENAVB osalTime] INFO: ml_phoffset = 217191007, ls_phoffset = 2912851934
[1545662387:160507080 OPENAVB osalTime] INFO: ml_freqffset = 0.999997, ls_freqoffset = 0.999968
[1545662387:160509154 OPENAVB Endpoint] INFO: AVTP Pipeline: 0.1.3 (Development)
[1545662387:162477606 OPENAVB Raw Socket] **ERROR: Checking interface; ioctl(SIOCGIFFLAGS) failed: No such device**
[1545662387:162494267 OPENAVB Endpoint] **ERROR: Invalid value: section=network, name=ifname, value=eth0**
[1545662387:162513102 OPENAVB Endpoint] **ERROR: Error in INI file: endpoint.ini, line 4**
[1545662387:162573351 OPENAVB Talker / Listener] INFO: AVTP Pipeline: 0.1.3 (Development)
[1545662387:162661099 OPENAVB Raw Socket] INFO: Using *pcap* implementation
[1545662387:263136855 OPENAVB Endpoint] **ERROR: Failed to initialize SRP**
[1545662387:343379350 OPENAVB Talker / Listener] INFO: Looking up symbol for function: openavbMapUncmpAudioInitialize
[1545662387:343395175 OPENAVB Talker / Listener] INFO: Looking up symbol for function: openavbIntfAlsaInitialize
[1545662387:343404718 OPENAVB 61883-6 Mapping] INFO: Rate:48000
[1545662387:343405942 OPENAVB 61883-6 Mapping] INFO: Bits:24
[1545662387:343406590 OPENAVB 61883-6 Mapping] INFO: Channels:8
[1545662387:343407311 OPENAVB 61883-6 Mapping] INFO: Packet Interval:8000
[1545662387:343407987 OPENAVB 61883-6 Mapping] INFO: Frames per packet:6
[1545662387:343408631 OPENAVB 61883-6 Mapping] INFO: Packing Factor:1
[1545662387:343409199 OPENAVB 61883-6 Mapping] INFO: Frames per MediaQ Item:8
[1545662387:343409882 OPENAVB 61883-6 Mapping] INFO: Sample Size Bytes:4
[1545662387:343566344 OPENAVB Talker / Listener] INFO: Looking up symbol for function: openavbMapUncmpAudioInitialize
[1545662387:343569782 OPENAVB Talker / Listener] INFO: Looking up symbol for function: openavbIntfAlsaInitialize
[1545662387:343574456 OPENAVB 61883-6 Mapping] INFO: Rate:48000
[1545662387:343575579 OPENAVB 61883-6 Mapping] INFO: Bits:24
[1545662387:343576154 OPENAVB 61883-6 Mapping] INFO: Channels:8
[1545662387:343576727 OPENAVB 61883-6 Mapping] INFO: Packet Interval:8000
[1545662387:343577291 OPENAVB 61883-6 Mapping] INFO: Frames per packet:6
[1545662387:343577920 OPENAVB 61883-6 Mapping] INFO: Packing Factor:256
[1545662387:343578443 OPENAVB 61883-6 Mapping] INFO: Frames per MediaQ Item:1536
[1545662387:343578970 OPENAVB 61883-6 Mapping] INFO: Sample Size Bytes:4
[1545662389:442629251 OPENAVB Endpoint MAAP] INFO: MAAP address range 38 acquired:  0x91e0f0004076-0x91e0f0004095 (Size 32)
[1545662389:442730944 OPENAVB Endpoint] **ERROR: Make sure that mrpd daemon is started**.
[1545662390:543417090 OPENAVB Endpoint] **ERROR: Failed to initialize SRP**
[1545662392:637589776 OPENAVB Endpoint MAAP] INFO: MAAP address range 39 acquired:  0x91e0f000c3cf-0x91e0f000c3ee (Size 32)
[1545662392:637690557 OPENAVB Endpoint] ERROR: Make sure that mrpd daemon is started.
[1545662393:738218800 OPENAVB Endpoint] ERROR: Failed to initialize SRP

I was namaged to remove the first error (eth0 not found) by manualy editing the .ini files in /build/bin (but I couldn't find a way to make it not built this way...). 

Regarding the second error- any idea why the SRP failes? 
mrpd seems to be working fine...

I also tried to build this with AVB_FEATURE_ENDPOINT=0. I didn't get the errors, but no data seemed to be sent, which was weird. 

(BTW- the "make avtp_pipeline_clean" seemed to delete the openavb_avdecc folder from the avtp_pipeline, and it is not rebuilt when I use "make avtp_pipeline" again )

So my questions are: 
1. Should I use AVB_FEATURE_ENDPOINT=0 or not? If yes- why? Did that cause the problem?
2. Is there a way to edit the default eth0 in the .ini files without editing each one manually?
3. Any idea what happened to openavb_avdecc? 
4. Is it ok to use this system without a switch?
5. Any more tips how to make it work? :)

 Thanks!

JXter, 2018-12-31:
You need to change the interface name in endpoint.ini file to the one you are using, if you want to use endpoint. Setting AVB_FEATURE_ENDPOINT flag will enable the use of SRP, MAAP etc.

JXter, 2018-12-31:
If you are connecting talker and listener directly (without switch in between), you can go with AVB_FEATURE_ENDPOINT=0 configuration. In this configuration, you need to set the configuration parameters in ini files correctly in both talker and listener. Without endpoint, you will not be able to communicate events on stream between talker and listener.

In case of AVB_FEATURE_ENDPOINT=1, your interface name in endpoint.ini doesn't match with the interface you are using, because of which you are getting those errors.

shirArg, 2018-12-31:
OK, I checked it- I already changed the interface name in endpoint.ini file, but (I think) it is changed every time I rebuild the project, I asked if there was any way to change it constantly. 

What does AVB_FEATURE_ENDPOINT stands for? what does it decide?

Thanks

shirArg, 2019-01-02:
Update- it did work with a switch :) 
Thanks

pinealservo, 2019-01-02:
AVB_FEATURE_ENDPOINT enables a chunk of functionality to manage other protocols aside from AVTP. Building without it is useful for non-standard uses of the streaming protocols for testing purposes or when you are using other tools to manage them, but it's required for full functionality.

The build process will overwrite a bunch of the ini files in the build directory, but you can always copy the ones you're using to an install directory and run from there so you don't have to re-edit them all the time.

shirArg, 2019-01-03:
Thanks Everyone! 

(this can be closed now)

closed 2019-01-07 by pinealservo

chuanqima, 2019-07-08:
Hi shirArg,
     I'd like to ask you how you solved this problem. Does the avtp_pipe you compiled work with endpoint files? Now I have a problem, which should be the same as yours, but I have no clue. I hope to get your help, and I am very grateful.
[image]

shirArg, 2019-07-08:
Hi chuanqima- I'm sorry but I can't remember anything I did besides what is written in this thread. I think the solution really was to use an AVB enabled switch (I used Extreme Summit X430-8p). Sorry I can't give any further help...

chuanqima, 2019-07-08:
hi，shirArg,Thank you for your reply. My current situation is that two PCs are directly connected to the environment and I have not used the AVB enabled switch. I solved this problem now, I have some configuration in my endpoint.ini file, and I am running commands on two PCs now, and I can be notified by the avb protocol on the receiver side, but I am confused. There is no sound. Can you give me a direction to try? Thank you.

chuanqima, 2019-07-11:
> If you are connecting talker and listener directly (without switch in between), you can go with AVB_FEATURE_ENDPOINT=0 configuration. In this configuration, you need to set the configuration [...]
Hi，JXter
    My test environment is directly connected to two PCs using I210 network card. When compiling, use AVB_FEATURE_ENDPOINT=0, and then implement the 61883-6 example provided in the README.md file in avtp_pipeline under the lib folder. It is completely ok. Play a wav audio file on the talker side and listen on the listener side. But when I try to use AVB_FEATURE_ENDPOINT=1 (modify the interface name in the relevant .ini file), I can't achieve the desired effect when running the example of 61883-6. Why is this? (I wonder if I can modify the file configuration to achieve the purpose of this example?)
Any help from you is very meaningful, thank you. look forward to your reply.