Avnu/OpenAvnu #856: Avtp_pipeline- wav_file_talker works only with AVB_FEATURE_ENDPOINT=0

shirArg, 2019-01-03:
Hi, 
I'm trying to use the avtp pipeline to send a wav file from one computer to the other. 
Both computers have I210 nics, and they are connected using a AVB compatible switch (Extreme Summit x430 8p). 

After a while, I managed to make it work using AVB_FEATURE_ENDPOINT=0. 
Now I want to use AVB_FEATURE_ENDPOINT=1, but It doesn't work for me. 

In the talker computer I use: 
sudo ./openavb_harness -I $IFNAME -s $STREAMS -d 0 -a a0:36:9f:2d:01:ad wav_file_talker.ini,sr_class=$CLASS,map_nv_tx_rate=$RATE,max_transit_usec=$TRANSIT_USEC,intf_nv_file_name=Dont_Stop_Me_Now_EDITED.wav,report_seconds=$REPORT

I get no errors in the log (attached in the end). 
My switch is configured to enable AVB, and everything seems to be fine. 
But- I can't see IEEE-1722 packets sent from the computer (as oppose to using it with AVB_FEATURE_ENDPOINT=0, which enabled me to see packets). 

Any idea why this is not working? Anything else I need to do?....

Thank you very much!

My log: 

sudo ./openavb_harness -I $IFNAME -s $STREAMS -d 0 -a a0:36:9f:2d:01:ad wav_file_talker.ini,sr_class=$CLASS,map_nv_tx_rate=$RATE,max_transit_usec=$TRANSIT_USEC,intf_nv_file_name=Dont_Stop_Me_Now_EDITED.wav,report_seconds=$REPORT
Opening: wav_file_talker.ini,sr_class=A,map_nv_tx_rate=8000,max_transit_usec=2000,intf_nv_file_name=Dont_Stop_Me_Now_EDITED.wav,report_seconds=0,dest_addr=91:e0:f0:00:fe:00,stream_addr=a0:36:9f:2d:01:ad,stream_uid=0,ifname=enp4s0
Opening: wav_file_talker.ini,sr_class=A,map_nv_tx_rate=8000,max_transit_usec=2000,intf_nv_file_name=Dont_Stop_Me_Now_EDITED.wav,report_seconds=0,dest_addr=91:e0:f0:00:fe:01,stream_addr=a0:36:9f:2d:01:ad,stream_uid=1,ifname=enp4s0
Opening: wav_file_talker.ini,sr_class=A,map_nv_tx_rate=8000,max_transit_usec=2000,intf_nv_file_name=Dont_Stop_Me_Now_EDITED.wav,report_seconds=0,dest_addr=91:e0:f0:00:fe:02,stream_addr=a0:36:9f:2d:01:ad,stream_uid=2,ifname=enp4s0
Opening: wav_file_talker.ini,sr_class=A,map_nv_tx_rate=8000,max_transit_usec=2000,intf_nv_file_name=Dont_Stop_Me_Now_EDITED.wav,report_seconds=0,dest_addr=91:e0:f0:00:fe:03,stream_addr=a0:36:9f:2d:01:ad,stream_uid=3,ifname=enp4s0
Opening: wav_file_talker.ini,sr_class=A,map_nv_tx_rate=8000,max_transit_usec=2000,intf_nv_file_name=Dont_Stop_Me_Now_EDITED.wav,report_seconds=0,dest_addr=91:e0:f0:00:fe:04,stream_addr=a0:36:9f:2d:01:ad,stream_uid=4,ifname=enp4s0
Opening: wav_file_talker.ini,sr_class=A,map_nv_tx_rate=8000,max_transit_usec=2000,intf_nv_file_name=Dont_Stop_Me_Now_EDITED.wav,report_seconds=0,dest_addr=91:e0:f0:00:fe:05,stream_addr=a0:36:9f:2d:01:ad,stream_uid=5,ifname=enp4s0
Opening: wav_file_talker.ini,sr_class=A,map_nv_tx_rate=8000,max_transit_usec=2000,intf_nv_file_name=Dont_Stop_Me_Now_EDITED.wav,report_seconds=0,dest_addr=91:e0:f0:00:fe:06,stream_addr=a0:36:9f:2d:01:ad,stream_uid=6,ifname=enp4s0
Configuring: wav_file_talker.ini,sr_class=A,map_nv_tx_rate=8000,max_transit_usec=2000,intf_nv_file_name=Dont_Stop_Me_Now_EDITED.wav,report_seconds=0,dest_addr=91:e0:f0:00:fe:00,stream_addr=a0:36:9f:2d:01:ad,stream_uid=0,ifname=enp4s0
Configuring: wav_file_talker.ini,sr_class=A,map_nv_tx_rate=8000,max_transit_usec=2000,intf_nv_file_name=Dont_Stop_Me_Now_EDITED.wav,report_seconds=0,dest_addr=91:e0:f0:00:fe:01,stream_addr=a0:36:9f:2d:01:ad,stream_uid=1,ifname=enp4s0
Configuring: wav_file_talker.ini,sr_class=A,map_nv_tx_rate=8000,max_transit_usec=2000,intf_nv_file_name=Dont_Stop_Me_Now_EDITED.wav,report_seconds=0,dest_addr=91:e0:f0:00:fe:02,stream_addr=a0:36:9f:2d:01:ad,stream_uid=2,ifname=enp4s0
Configuring: wav_file_talker.ini,sr_class=A,map_nv_tx_rate=8000,max_transit_usec=2000,intf_nv_file_name=Dont_Stop_Me_Now_EDITED.wav,report_seconds=0,dest_addr=91:e0:f0:00:fe:03,stream_addr=a0:36:9f:2d:01:ad,stream_uid=3,ifname=enp4s0
Configuring: wav_file_talker.ini,sr_class=A,map_nv_tx_rate=8000,max_transit_usec=2000,intf_nv_file_name=Dont_Stop_Me_Now_EDITED.wav,report_seconds=0,dest_addr=91:e0:f0:00:fe:04,stream_addr=a0:36:9f:2d:01:ad,stream_uid=4,ifname=enp4s0
Configuring: wav_file_talker.ini,sr_class=A,map_nv_tx_rate=8000,max_transit_usec=2000,intf_nv_file_name=Dont_Stop_Me_Now_EDITED.wav,report_seconds=0,dest_addr=91:e0:f0:00:fe:05,stream_addr=a0:36:9f:2d:01:ad,stream_uid=5,ifname=enp4s0
Configuring: wav_file_talker.ini,sr_class=A,map_nv_tx_rate=8000,max_transit_usec=2000,intf_nv_file_name=Dont_Stop_Me_Now_EDITED.wav,report_seconds=0,dest_addr=91:e0:f0:00:fe:06,stream_addr=a0:36:9f:2d:01:ad,stream_uid=6,ifname=enp4s0
Starting: wav_file_talker.ini,sr_class=A,map_nv_tx_rate=8000,max_transit_usec=2000,intf_nv_file_name=Dont_Stop_Me_Now_EDITED.wav,report_seconds=0,dest_addr=91:e0:f0:00:fe:00,stream_addr=a0:36:9f:2d:01:ad,stream_uid=0,ifname=enp4s0
Starting: wav_file_talker.ini,sr_class=A,map_nv_tx_rate=8000,max_transit_usec=2000,intf_nv_file_name=Dont_Stop_Me_Now_EDITED.wav,report_seconds=0,dest_addr=91:e0:f0:00:fe:01,stream_addr=a0:36:9f:2d:01:ad,stream_uid=1,ifname=enp4s0
Starting: wav_file_talker.ini,sr_class=A,map_nv_tx_rate=8000,max_transit_usec=2000,intf_nv_file_name=Dont_Stop_Me_Now_EDITED.wav,report_seconds=0,dest_addr=91:e0:f0:00:fe:02,stream_addr=a0:36:9f:2d:01:ad,stream_uid=2,ifname=enp4s0
Starting: wav_file_talker.ini,sr_class=A,map_nv_tx_rate=8000,max_transit_usec=2000,intf_nv_file_name=Dont_Stop_Me_Now_EDITED.wav,report_seconds=0,dest_addr=91:e0:f0:00:fe:03,stream_addr=a0:36:9f:2d:01:ad,stream_uid=3,ifname=enp4s0
Starting: wav_file_talker.ini,sr_class=A,map_nv_tx_rate=8000,max_transit_usec=2000,intf_nv_file_name=Dont_Stop_Me_Now_EDITED.wav,report_seconds=0,dest_addr=91:e0:f0:00:fe:04,stream_addr=a0:36:9f:2d:01:ad,stream_uid=4,ifname=enp4s0
Starting: wav_file_talker.ini,sr_class=A,map_nv_tx_rate=8000,max_transit_usec=2000,intf_nv_file_name=Dont_Stop_Me_Now_EDITED.wav,report_seconds=0,dest_addr=91:e0:f0:00:fe:05,stream_addr=a0:36:9f:2d:01:ad,stream_uid=5,ifname=enp4s0
Starting: wav_file_talker.ini,sr_class=A,map_nv_tx_rate=8000,max_transit_usec=2000,intf_nv_file_name=Dont_Stop_Me_Now_EDITED.wav,report_seconds=0,dest_addr=91:e0:f0:00:fe:06,stream_addr=a0:36:9f:2d:01:ad,stream_uid=6,ifname=enp4s0
[1546524069:215223718 OPENAVB osalTime] INFO: local_time = 1546524068991161711
[1546524069:215231210 OPENAVB osalTime] INFO: ml_phoffset = -615068664021, ls_phoffset = 210779057
[1546524069:215237675 OPENAVB osalTime] INFO: ml_freqffset = 1.000044, ls_freqoffset = 0.999967
[1546524069:215239697 OPENAVB Endpoint] INFO: AVTP Pipeline: 0.1.3 (Development)
[1546524069:215339908 OPENAVB Talker / Listener] INFO: AVTP Pipeline: 0.1.3 (Development)
[1546524069:215496833 OPENAVB Talker / Listener] INFO: Looking up symbol for function: openavbMapUncmpAudioInitialize
[1546524069:215508314 OPENAVB Talker / Listener] INFO: Looking up symbol for function: openavbIntfWavFileInitialize
[1546524069:215540984 OPENAVB Wav File Interface] INFO: Number of data bytes:38285312
[1546524069:215544398 OPENAVB Wav File Interface] INFO: Wav file - Rate:44100 Bits:16 Channels:2
[1546524069:215689849 OPENAVB Wav File Interface] INFO: Number of data bytes:38285312
[1546524069:215693731 OPENAVB Wav File Interface] INFO: Wav file - Rate:44100 Bits:16 Channels:2
[1546524069:215696191 OPENAVB 61883-6 Mapping] WARNING: audio rate (44100) is not an integer multiple of TX rate (8000). Recommend TX rate of (7350)
[1546524069:215697764 OPENAVB 61883-6 Mapping] INFO: Rate:44100
[1546524069:215698748 OPENAVB 61883-6 Mapping] INFO: Bits:16
[1546524069:215699798 OPENAVB 61883-6 Mapping] INFO: Channels:2
[1546524069:215700755 OPENAVB 61883-6 Mapping] INFO: Packet Interval:8000
[1546524069:215701664 OPENAVB 61883-6 Mapping] INFO: Frames per packet:6
[1546524069:215702548 OPENAVB 61883-6 Mapping] INFO: Packing Factor:32
[1546524069:215703734 OPENAVB 61883-6 Mapping] INFO: Frames per MediaQ Item:192
[1546524069:215704656 OPENAVB 61883-6 Mapping] INFO: Sample Size Bytes:4
[1546524069:215882141 OPENAVB Talker / Listener] INFO: Looking up symbol for function: openavbMapUncmpAudioInitialize
[1546524069:215951818 OPENAVB Talker / Listener] INFO: Looking up symbol for function: openavbIntfWavFileInitialize
[1546524069:215978686 OPENAVB Wav File Interface] INFO: Number of data bytes:38285312
[1546524069:215980887 OPENAVB Wav File Interface] INFO: Wav file - Rate:44100 Bits:16 Channels:2
[1546524069:215988399 OPENAVB Wav File Interface] INFO: Number of data bytes:38285312
[1546524069:215989469 OPENAVB Wav File Interface] INFO: Wav file - Rate:44100 Bits:16 Channels:2
[... cut ...]
[1546524323:934396866 OPENAVB Endpoint] INFO: SRP talker callback uid=2: lsnrDecl=1
[1546524323:934411744 OPENAVB Endpoint] INFO: SRP talker callback uid=3: lsnrDecl=1
[1546524323:934425897 OPENAVB Endpoint] INFO: SRP talker callback uid=4: lsnrDecl=1
[1546524323:934440985 OPENAVB Endpoint] INFO: SRP talker callback uid=5: lsnrDecl=1
[1546524323:934455449 OPENAVB Endpoint] INFO: SRP talker callback uid=6: lsnrDecl=1
[1546524327:523799842 OPENAVB Endpoint] INFO: SRP talker callback uid=0: lsnrDecl=1

andrew-elder, 2019-01-03:
When you say you can't "see" any packets, what do you mean exactly? Do you have a network TAP in-line with the i210? Or, some other method?

shirArg, 2019-01-03:
Sorry! I meant I don't see them in Wireshark (and I did see them when using it with AVB_FEATURE_ENDPOINT=0)

andrew-elder, 2019-01-03:
Wireshark connected to what exactly? The i210? A second NIC connected to a network TAP?

shirArg, 2019-01-03:
My setup is: 
PC1 <-> I210 <-> Switch <-> I210 <-> Pc2
The Wireshark sniffs the I210 of the first computer, that's where I expect to see the packets. 
No tap is used, and I don't use the PC's built in nic at all. 

The weird thing is that (after a lot of work) it worked perfectly with AVB_FEATURE_ENDPOINT=0, but now no audio packets are even sent...

Thanks

andrew-elder, 2019-01-03:
I'm not too sure of the current state (someone else probably has more details), but it used to be that the talker packets were sent to the i210 from a sort of backdoor descriptor path - so they wouldn't show up in the regular linux network stack (and therefore wireshark).

shirArg, 2019-01-03:
I know what you are talking about- as far as I seen this is what happening in the simple_talker example. 

I was assuming this is not what's going on in avtp_pipeline, especially after seeing the packets in Wireshark when it did work (using AVB_FEATURE_ENDPOINT=0). I guess the sending mechanism is not changed when changing AVB_FEATURE_ENDPOINT (but I didn't look into the code, so maybe I'm wrong...). Anyway- I can't see packets nor hear music in the other PC too. 

Can you think of anything else I can do to investigate this? It's kind of tricky since the logs seems fine (maybe expect this rate warning, but I don't assume this is what it all about, and again it worked in the simpler case with the same song...)

Thanks!

shirArg, 2019-01-03:
I managed to make it work!!

So the problem was actually in the switch configuration. Searching through the configuration I found this- 
[image]

The problem was that I configured the switch to use VLAN 2 (in the previous level, I saw in Wireshark that the packets are sent with vlan tag 2, but don't pass the switch), but I didn't configure it to be tagged, and for some reason it messed up the SRP protocol. The solution was to tell the switch to use tags (in this switch - `configure vlan "Default" add ports 1-10 tagged'), so after this the result of the msrp was- 

[image]

And suddenly everything works :) (I can see the packets in Wireshark, and hear the song in the other PC). 

I don't really know what was the problem in the avtp code, but it seems that there should be somethings that understands the SRP is not really working, and tells the user about it... 

Thank you very much for the help! (And I hope someone else will find it useful in the future...)