Avnu/OpenAvnu #857: Received PDelay FUP has different seqID than the PDelay request

DeepSwimmingShark, 2019-01-14:
Hi Everyone,
I was executing my gPTP deamon for the server and the client . I could see that the AsCapable: Enabled in both the system but still i get the following error "Received PDelay FUP has different seqID than the PDelay request". 

This might have happened because the Pdelay Response follow up message came earlier than the Pdelay Request message or it might have come too late that already another Pdelay Request message have triggered in between. As a result the sequence id of both is not matching.

I checked with the wireshark and found that there is no unexpected Pdelay Request interfering in between a Pdelay Request and Pdelay Response Follow up message. In that case, why does my Sequence id not matching? and what should I do to solve this error?

I have attched the images of gPTP executed on my server and client. Please do help me.
TSN Server:
[image]

TSN Client:
[image]

Thanks in advance!

christopher-s-hall, 2019-01-14:
If the followup sequence was early or late I would expect an off by 1 error. Make sure that there is only one instance of the gPTP daemon running on both server and client. How are the client/server connected? They should be directly connected. Which OS/version/hardware are you using?

DeepSwimmingShark, 2019-01-15:
I have connected my two PC’s having intel I210 external network interface card directly with a cross over ethernet. I have not used any switches in between.And I am running only one instance of deamon on both the server and client. I am running this deamon on Linux OS  Version:4.15.0-43-generic.

But even after stopping the deamon on both the PC, I could see that there is still Peer delay request message being requested and I think its normal. I have attached the packets captured by the wireshark in server and client. My gPTP deamon considers my grandmaster as the server system with the MAC address 68:05:CA:FF:FE:8B:0D:CD. Please have a look on it and let me know if you could identify any mistake in it.

Server (enp1s0):
[image]

Client (enp2s0):

[image]

DeepSwimmingShark, 2019-01-15:
The one error I could identify on my side is that each time I run the deamon on the server, the sequence Id of follow up doesnt starts counting from 1 but instead it continues somewhere near the previous value the deamon was stopped. And the sequence ID of peer delay request  starts counting from "1" every time I run the deamon. This might be the mistake why sequence ID is not matching. But does anyone have an idea how to make the sequence id of follow  up start from "1".
I will show you some instances that i ran on the server. Please note that I ran the next instances on both the PC's after stopping all the previous instances. You can see the error directly:
**Server:**
First instance:
[image]

Second instance:
[image]

Third instance:

[image]

 It might be because the script is still running in background. But i tried stopping the script by both the ways "Ctrl +C" and"Ctrl +Z". And I also tried killing all the process by "kill -9" and restarted the deamon but still the same error exists.

ClickB, 2020-03-23:
Hi EmbFrillShar, Have you solved thi seq question? Can you share it for me？I have the same question like you.

groenvit, 2020-05-12:
I had the same problem. 
Since no solution has been provided here yet, I will briefly describe my cause.
It turns out that christopher-s-hall's hunch was right.
In my case it was due to a double occupancy.

andrew-elder, 2020-05-12:
Thanks @groenvit. Since a solution has been provided and confirmed, I'm going to close this issue.

closed 2020-05-12 by andrew-elder