Avnu/OpenAvnu #858: Non igb rawsockets not using avb?

JXter, 2019-02-01:
Hi,
I was experimenting on the OpenAVNU codebase by trying to send more data than reserved. For that purpose I have disabled the sleep functions in talker, increased transmit rate, and tweaked interface module to give some data everytime when callback is called. My idea was to see that, even if I try to send more data through any rawsockets, device will send data only according to the reservation. I observed that with igb rawsockets, it is limiting the packets send, but for pcap or simple rawsockets, there is no restriction.  Any idea, what might be going wrong?

pinealservo, 2019-02-02:
The idea with the non-igb usage is to use the native kernel stack for traffic shaping, which can be configured via the `tc` command. The shaper daemon in the repository currently configures kernel software shapers into an approximation that should work for a stream or two. Recent kernels have better AVB traffic shaping support (using some newer shaping modules that should configure the i210 shaping when available but fall back to software if not) but the shaper daemon hasn't been updated to use that new method. As far as I know, there are no plans for anyone to work on that, so it would be a great project for someone to contribute....

JXter, 2019-02-04:
Thank you @pinealservo.