Avnu/OpenAvnu #859: Add interface module to avtp pipeline

chris-kuhr, 2019-02-07:
Hi *,

I want to write a JACK interface module for the avtp pipeline.
The alsa interface serves as a template, I added the JACK build parameters in (nearly) all relevant CMakeLists and created an .ini file.

The compilation works, but the openavb_host still shows the symbols of the ALSA interface.

What do I have to edit in order to compile the executable correctly?

BR,
Ck

chris-kuhr, 2019-04-17:
hi,

can anyone help?
I am stuck and am not able to find the missing link...

Best,
Ck

JakubMielczarek81, 2019-04-18:
Hi,

You need to register the interface module in openavb_host and harness by calling "registerStaticIntfModule()". For example:

registerStaticIntfModule(openavbIntfEchoInitialize);

Have a look at host and harness and how they register the existing interfaces and mappings.

Best regards,
Jakub

chris-kuhr, 2019-04-18:
Jakub, 

thank you very much! This was the last missing link.

Now everything looks fine.

Best,
Ck

closed 2019-04-18 by andrew-elder