Avnu/OpenAvnu #861: How to sync two gPTP instances?

sshydlovskyi, 2019-04-16:
Hi!

I'm trying to get two gPTP daemon instances running in sync on two Raspberry Pi boards to check sync signal on GPIO pins. Two boards are on the same Wi-Fi network and can ping each other.

When I run one of them with -T command line switch to force it to be master and another one with -L switch to be a slave, the second one after short time interval says that after waiting for sync it becomes master too. I also add -S switch to start synchronization to both commands.

Can I achieve a master-slave connection in this case? If yes, what else am I missing? Maybe, some other command switches or something else?

Thanks.

pinealservo, 2019-04-16:
This is not a configuration that will work with standard gPTP, which typically requires hardware timestamping that's not available in the RaspberryPi.  There's a branch, https://github.com/AVnu/OpenAvnu/tree/feature-aptp-experimental, which adds an Apple-specific profile that is designed to work on Raspberry Pi with WiFi (and presumably interop with Apple Airplay2 devices), but I can't tell you much about how well it works or how to get it working.

sshydlovskyi, 2019-04-16:
> This is not a configuration that will work with standard gPTP, which typically requires hardware timestamping that's not available in the RaspberryPi. There's a branch, [...]

Thank you for your reply!

Moreover, I have found a separate fork, https://github.com/rroussel/OpenAvnu, which have PTP daemon version with Apple profile and only with that fork I could get the build that works on RPi. Unfortunately, there is no issues section, so can't ask there.

Thanks anyway, will check.

chennakeshavaSoC, 2019-08-12:
I am also trying to run the AVB on RPi and facing challenges.

@sshydlovskyi how did you managed to run on RPi?
are you able to test the PTP using rroussel code? I will try the same, please let me know if any modifications to be made to work with Pi.

sshydlovskyi, 2019-08-28:
@chennakeshavaSoC Hi! Sorry for the delay with the answer. Yes, I think I used rrousel variant, just need to check all build options. Will write you back.