Avnu/OpenAvnu #864: INtel i210 + AVNU stackwith TI DRA7xx

RameshRex, 2019-05-24:
I plan to re-compile AVnu stack and drivers for TI board and use an INtel i210 for a quick demo.
If i have say 3 such devices, will i be able to demo TSN part in audio/ video  . I

pinealservo, 2019-05-26:
If by "quick demo" you mean something that doesn't require much work on your part to set it up, I would not bet on it. I don't know enough about your background or expertise to say how much work/time it would take, but the repository here is definitely a toolbox with "some assembly required" right now rather than a set of ready-to-run functional demo applications.

Some things to consider:

Most of the i210 TSN driver development is done and tested on Intel platforms; the i210-specific driver might work on an ARM kernel now, but at least at one point it did not. You'll have to verify it works on your endpoints.

You've mentioned 3 endpoints but not a switch. You will have to ensure you have a compliant switch set up properly and that your endpoints can successfully get gPTP sync and do SRP reservations over it.

Even with those out of the way, getting media synchronization working in a way that complies with AVTP generally requires careful selection of media codec hardware and especially the hardware configuration with respect to media clock measurement and adjustment. Unless you're using a board that's been designed with TSN in mind, it's unlikely to work well; attempting to dial down latency will just yield buffer over/under-runs due to media clock mismatches. You may also have difficulty scheduling stream starts at the correct presentation time.

Your board may very well have been designed for TSN though; in this case you'll need to understand how the hardware designers intended for the media sync to work and integrate any necessary driver changes to access that hardware. Few boards are set up to do this exactly the same way, so we haven't been able to provide a general purpose method of doing media synchronization. There are hooks for it in the avtp_pipeline library, but unfortunately there's no support or examples provided for implementation.

There's some work in avtp_pipeline that attempts to do all the scheduling in software with only gPTP timestamps; this can work reasonably well with one or two streams if you're willing to allow for a bit more latency and fairly coarse methods of software media rate matching and time synchronization. I wouldn't recommend using this without having some experience with compliant AVB/TSN hardware and the means to test it to see if you're really achieving what you think you are.

RameshRex, 2019-05-27:
Thank you very much for the valuable inputs!
My plan is to port the i210 driver for the ti jacinto 6 platform and use ttech switch for connecting the other render devices.

The to jacinto as talker and a couple other nodes on the switch as listeners.
Will try to use a supported media format as well for the avtp. 

Talker
Ti jacinto running avb stack,+ drivers - > Intel i210 hardware - > ttech switch

Listeners
Ttech switch->  listener mode with avb stack

pinealservo, 2019-05-27:
Is there some reason that you want to use the i210 with the Jacinto 6?  It has a built-in AVB-capable Ethernet subsystem, with the advantage that the timestamp system can also be internally hooked to the McASP audio ports. I think they may have a credit-based shaper implemented internally as well.

RameshRex, 2019-06-25:
Thanks for your inputs. Very helpful indeed since I'm new to OpenAVNU

closed 2019-06-25 by RameshRex