Avnu/OpenAvnu #867: Cannot build on Windows 10

rreed, 2019-06-13:
Possibly just me being dumb, and not knowing what to build, so first I'll start out by saying what I want to do: I want to run the example in `run_echo_listener`. Okay, cool:

```
./run_echo_listener.sh: line 10: cd: lib/avtp_pipeline/build/bin: No such file or directory
```

Fine, so I need to build `avtp_pipeline`, which, as I understand it, means that I need to do `make all`. I can generate the MinGW files just fine (`mkdir tmp + cd tmp + `cmake .. -G "MinGW Makefiles"`)

```
-- Generating done
-- Build files have been written to: C:/Users/RR/workspace/openavnu/tmp
```

Now I can go in two directions, both of which fail. First, running `make all` from inside the new cmake tmp directory:

```
C:\Users\RR\workspace\openavnu\tmp>make
[  0%] Building CXX object thirdparty/cpputest/src/CppUTest/CMakeFiles/CppUTest.dir/CommandLineArguments.cpp.obj
In file included from c:\mingw\lib\gcc\mingw32\8.2.0\include\c++\cstdio:42,
                 from c:\mingw\lib\gcc\mingw32\8.2.0\include\c++\ext\string_conversions.h:43,
                 from c:\mingw\lib\gcc\mingw32\8.2.0\include\c++\bits\basic_string.h:6391,
                 from c:\mingw\lib\gcc\mingw32\8.2.0\include\c++\string:52,
                 from c:\mingw\lib\gcc\mingw32\8.2.0\include\c++\stdexcept:39,
                 from c:\mingw\lib\gcc\mingw32\8.2.0\include\c++\array:39,
                 from c:\mingw\lib\gcc\mingw32\8.2.0\include\c++\tuple:39,
                 from c:\mingw\lib\gcc\mingw32\8.2.0\include\c++\bits\unique_ptr.h:37,
                 from c:\mingw\lib\gcc\mingw32\8.2.0\include\c++\memory:80,
                 from C:/Users/RR/workspace/openavnu/thirdparty/cpputest/include/CppUTest/MemoryLeakDetectorNewMacros.h:47,
                 from <command-line>:
c:\mingw\include\stdio.h:788:34: error: '__off64_t' does not name a type; did you mean '__time64_t'?
```

And just trying to create `avtp_pipeline/build` from the Makefile one level up so I can run the original thing I wanted to:

```
C:\Users\RR\workspace\openavnu>make avtp_pipeline
mkdir -p lib/igb_avb/lib && make  subdir=lib/igb_avb/lib  -C lib/igb_avb/lib
The syntax of the command is incorrect.
Makefile:36: recipe for target 'lib' failed
```

So uh, what do I do here?

andrew-elder, 2019-06-13:
I think most folks working on AVB are building in Linux. I'm not surprised there is something not working in the migw environment. Do you have a Linux machine you can test on?

rreed, 2019-06-13:
I do, but the application I need to eventually run this with is Windows-only, because customers. So getting it to run in Linux doesn't help me. :)

jdkoftinoff, 2019-06-13:
Get it working on linux as a baseline (not in a virtual machine). This allows you to learn all about how it works and how to test it. Then on windows you would not use mingw32 ideally.

andrew-elder, 2019-06-13:
Getting it all to work in Windows will be challenging. Have a look here http://www.audioscience.com/internet/news/press_Hono_VSC_Basic.htm.

ahogen, 2019-07-10:
Shameless plug :wink:

zarfld, 2020-06-29:
unfortunately "Hono VSC Basic" is not available any more for purchase. 

When trying to compile it - at least gPtP & MRPD seem somehow to basically run, when i used vs2019 to compile it.
(iirc setting up WPCAP_DIR was the most important change )

 but i doubt that maap does:

e.g. in maap_main.c
> 
/*
 * TODO:  This code still needs to be added!
 */
>