Avnu/OpenAvnu #870: how to run avb on ubuntu（with i210） and demoboard （support avb）

chuanqima, 2019-06-27:
Dear Sir,
I am currently trying to use a Linux PC with an I210 network card to communicate with a hardware avb bridge, simulating the environment where the development board and PC synchronize clocks. Is there any difference between this environment and the one where the two PCS are directly connected?I don't know much about avb, and I just started to study it recently. I hope to get your advice, where can I start?I've built all the projects, and now I can synchronize the time between the two development boards, and I can implement the speaker and listener examples, but I don't have a clue about the connection between PC and development board, I hope to get your answer, I will be very grateful.Thank you ^ - ^

andrew-elder, 2019-06-27:
I don't understand your question. Do you mean AVB stream connection to a development board of some kind? Does the development board have a complete AVB software stack running on it, including 1722.1? If you can provide more detail that may help.

chuanqima, 2019-06-27:
Yes, Sir, I have a stack on my development board to run the full avb software, and before setting up this environment I was able to complete both boards by running the openavb_across-i eth0-e /data/avb/ endo.ini /data/avb/alsa2_talker.ini /data/avb/alsa2_listener.ini command to speak and listen.

chuanqima, 2019-06-27:
and When two boards communicate, one is the master and the other is the slave,

andrew-elder, 2019-06-27:
gPTP has a master/slave concept. The AVTP packets have a talker/listener concept. So you are saying you can send audio packets ok?

chuanqima, 2019-06-27:
Yeah, but now I want to send an audio packet between the board and the computer, and at the very least implement talker and listener. I don't know what I should do on the computer side.Any examples?Is the premise of communication that gPtp can synchronize the clock first?

chuanqima, 2019-06-27:
and sir, before i running openavb_harness -I eth0 -e /data/avb/endpoint.ini /data/avb/alsa2_talker.ini /data/avb/alsa2_listener.ini command ,i modify the avb configuration: ln /avb/endpoint.ini set "preconfigure = 1", ln /avb/alsa2_listener.ini set "stream_addr" and "dest_addr", ln /avb/alsa2_talker.ini set "stream_addr" and "dest_addr", and the two add are mac add for each board.so i can build two channels,one master-slave listen/talk one is slave-master talk/listen. and can i do the same thing between the PC and the board ? if not, what should i do.

chuanqima, 2019-07-01:
Mr. Andrew，
Can you help me answer some of the following questions?
Now is my development board and the computer is directly connected to the computer through the network interface converter. On the computer is the i210 network card. His role in both is from the computer is the master. When I open the daemon_cl process, they should be connected when I see the printed information, as shown in the following figure.
[image]

1. Why can't I run the run_daemons.sh script directly in the root directory on the computer side (master)? Because my development board(slave) doesn't seem to change roles at this time. But if I run daemons/gptp/linux/build/obj/daemon_cl enp1s0-F daemons/gptp/gptp_cfg.ini on the pc(master), slave will change  role from master to slave. Why? Because of some configurations in gptp_cfg.ini?

2. I wondered why the message I saw through Wireshark was wrong. For example, why does the web ports on the development board send Resp messages? Why do computers send Req messages?
[image]

regard!

closed 2019-07-04 by chuanqima

chuanqima, 2019-07-09:
hi, elder,
I ran the eavb startup daemon script, run_daemon, on two NICs with I210, but I don't know if I want to rewrite the following script, especially if it starts in the script "daemons/maap/linux/build/ Maap_daemon -i $nic -d /dev/null "This command, do I need to change this command on the listener side? Change to "daemons/maap/linux/build/maap_daemon -c -d /dev/null
"What?"
In addition, I ran sudo ./openavb_harness -I enp1s0 wav_file_talker.ini, sr_class=A, map_nv_tx_rate=7350, max_transit_usec=2000, intf_nv_file_name=ringtone1.wav, report_seconds=0 command on the talker side.
Run sudo on the listener side. /openavb_harness -I enp4s0
I want to play ringtone1 audio on the child talker, and I can receive the audio at the listening end. The audio-initiated command sudo ./openavb_harness -I enp1s0 -a 6c:b3:11:52:3b:99 alsa_listener.ini,sr_class=A,map_nv_tx_rate=8000,max_transit_usec=2000,report_seconds=0. After the two commands are executed separately, there is no sound. Why is this? I modified the value of SRP on the endpoint, preconfigured = 1.

Also, when I executed these two commands, I killed the two commands manually because there was no sound for a long time. Found that there is an additional enpoint_save.ini file in lib/avtp_pipeline/build/bin. What does this file do?

Can you take the time to help me see this problem? Grateful.
发送反馈