Avnu/OpenAvnu #874: GPTP acCapable is enable ,but Error On 'Master'

chuanqima, 2019-07-05:
Hi, everybody.
Recently, I made an attempt to connect the embedded development board with PC (I210 network card) directly to test whether my development board can transplant open-avb normally. Here is my step:
1. Run daemon_cl on the development version, start gPTP, and the board will become the main state
2. Run daemon_cl on PC and start gPTP. (When I compile avtp_pipeline, I use'AVB_FEATURE_ENDPOINT=0 make avtp_pipeline'). At this point, my PC will display: AsCapable: Enabled and Switching to Master. And the development board will become Slave. At the same time: As Capable: Enabled. You can also see them sending'Path_Delay_Req_Message'to each other through the package grabbing tool. Can I think that the board and my PC can be synchronized in time?
3. But then there was the following error message on my PC (Master):
EXCEPTION: GPTP [16:58:47:454] Sync discontinuity
ERROR: GPTP [16:59:00:364] > > Received PDelay followup but no RESPONSE exists
ERROR: GPTP [16:59:04:431] > > Received PDelay followup but no RESPONSE exists
ERROR: GPTP [17:03:50:945]>Received PDelay followup but no RESPONSE exists
ERROR: GPTP [17:05:08:587]>Received PDelay followup but no RESPONSE exists
ERROR: GPTP [17:06:04:176]>Received PDelay followup but no RESPONSE exists
ERROR: GPTP [17:06:22:285]> Received PDelay followup but no RESPONSE exists
ERROR: GPTP [17:07:17:784]> Received PDelay followup but no RESPONSE exists
ERROR: GPTP [17:08:25:383]>Received PDelay followup but no RESPONSE exists
[image]

There are no errors on my development board！
Tks,

chuanqima, 2019-08-20:
@andrew-elder  Dear sir, please forgive my presumption. I know you have no obligation to answer this question for me, but I still hope to get some help from you. I can't thank you enough.

andrew-elder, 2019-08-20:
I would run wireshark on your Linux PC with the i210 and see if the packet traffic is what are expecting.

I would also double check that the network interface on the dev board is running as expected. Are you using an i210 on the dev board as well?

chuanqima, 2019-08-21:
Sir, I am using TC9560 Neutrino chip in the development board, which is completely avb compliant. I think I can treat it as an i210.

pinealservo, 2019-08-21:
Looked up some info on TC9560; looks like it's potentially a nice solution and it should be really AVB compliant with the right drivers/firmware. The details of how it works are apparently only available under NDA, so except for what can be inferred from this old driver dump (https://github.com/ToshibaAmericaElectronicComponents?tab=repositories) I don't have any way of knowing how you make it work today.

I think it's safe to say that you can't treat it as an i210, though. You'll have to talk to your Toshiba representative to get help with configuring it to work in an AVB-compliant mode and troubleshoot problems.