Avnu/OpenAvnu #883: Getting GPTP to run on Intel 82579

satishbabu2136, 2019-09-24:
I am trying to run the gptp daemon with 3 systems as listeners, but system stops the daemons with the different error messages. This gptp was working fine with 2 systems, when I start the third system, it goes to the different errors like., invalid received timestamp, follow up response message sequence ID mismatch and port identity mismatch.

Will it support for the ethernet controller, Intel 82579?

satishbabu2136, 2019-09-25:
But, if i ran with 2 systems on Intel 82579, it is working fine. If it doesn't support hardware timestamping then how it's running with 2 systems?

jdkoftinoff, 2019-09-26:
The timing accuracy is what is important. Without hardware timing accuracy you will always have troubles.

satishbabu2136, 2019-09-26:
Thanks for the information.

closed 2019-09-26 by satishbabu2136

pinealservo, 2019-09-26:
The Intel 82579 *does* support hardware timestamping, though.  But apparently there have been driver bugs related to PTP support on it in the past: https://lore.kernel.org/patchwork/patch/761667/

These ought to be fixed on recent kernels, but you may need to experimentally determine the right PHY delay parameters and such things. You may also want to try the LinuxPTP software; it may support that NIC better than the gPTP daemon here, which has been largely focused on working well with i210.

Note that although it has hardware timestamping, it does *not* have any other AVB/TSN features and the i210 is what you want if you want to go further than just time sync.

jdkoftinoff, 2019-09-27:
Oh, good point! For the longest time the hardware time stamping just wasn’t enabled at all.

To the original poster of the question: If you have 3 devices hooked up to the network, which switch are you using? If it is not 802.1AS enabled you will see failure when you connect 3 devices.

Jeff