Avnu/OpenAvnu #884: Run gPTP with wirelessport in linux

padmajamedaka, 2019-10-03:
We would like to run AVB through a wireless port in Linux. 
If I go through the code, the default implementation is like wireless port-->windows:: ethernet port-->Linux.

Wireless port supports in windows and Ethernet port supported in Linux.
 I tried to replace the initialization of ethernet port with wireless port, but getting build errors.

How can I run this code through a wireless port in Linux?

andrew-elder, 2019-10-03:
There is some wireless code here https://github.com/AVnu/OpenAvnu/tree/feature-aptp-experimental, but I'm not sure of the state of it . . .

swhan91, 2021-02-15:
@andrew-elder Hi, I'd like to run AVB through a wireless port in Linux, too.
https://github.com/AVnu/OpenAvnu/tree/feature-aptp-experimental
This branch failed CI and It doesn't seem to be under development.

I wonder how it's going. I'll be grateful if you give me an answer.