Avnu/OpenAvnu #886: Facing issue, when executing gPTP on more than 2 PCs(Not getting synchronization)

ram-nagapuri, 2019-10-04:
Hi All, 
   I am using the master branch of OpenAvnu. My initial aim is to get synchronization with the gPTP daemon so I can move on. I was executed gptp on 2 linux PCs(both are I210 and Ubuntu version 14.4), they are running fine (without any errors) and could able to see AsCapable: Enabled in both the system. And They keep exchanging PDelay messages and sync messages. They are in synchronization, Please see the below logs

**OpenAvnu-master$ sudo ./run_daemons.sh  eth1
Starting daemons on eth1
INFO     : GPTP [20:11:34:007] gPTP starting
ERROR    : GPTP [20:11:34:008] Using clock device: /dev/ptp0
STATUS   : GPTP [20:11:34:008] Starting PDelay
STATUS   : GPTP [20:11:34:008] Link Speed: 1000000 kb/sec
STATUS   : GPTP [20:11:34:042] AsCapable: Enabled
process_events()
STATUS   : GPTP [20:11:35:857] Switching to Slave   //and other one it is Master
STATUS   : GPTP [20:11:35:857] New Grandmaster "68:05:CA:FF:FE:8D:32:98" (previous "00:00:00:00:00:00:00:00")**

   the Actual problem is when I executing the gptp on 3rd machine (same configuration), Getting errors on three PCs(running only single instance of gptp/PC). These 3 PCs are not connected to any external network. these are connected through a Network switch(connected through the eth1 interfaces of each device). Getting multiple errors "Received PDelay FUP has different seqID than the PDelay request (226/149)", "Source port identity from PDelay Response/FUP differ",  ">>> Received PDelay followup but no RESPONSE exists" (And adjusted the link delays to resolve link delay errors)

**STATUS   : GPTP [20:42:36:523] *** Sync Timeout Expired - Becoming Master
STATUS   : GPTP [20:42:36:523] New Grandmaster "68:05:CA:FF:FE:92:F6:FD" (previous "68:05:CA:FF:FE:8D:32:98")
ERROR    : GPTP [20:42:36:699] Received PDelay FUP has different seqID than the PDelay request (226/149)
ERROR    : GPTP [20:42:37:215] Received PDelay FUP has different seqID than the PDelay request (190/149)
EXCEPTION: GPTP [20:42:37:485] PDelay Response Receipt Timeout
ERROR    : GPTP [20:42:37:506] Source port identity from PDelay Response/FUP differ
ERROR    : GPTP [20:42:37:506] >>> Received PDelay followup but no RESPONSE exists
ERROR    : GPTP [20:42:37:701] Received PDelay FUP has different seqID than the PDelay request (227/150)
ERROR    : GPTP [20:42:38:218] Received PDelay FUP has different seqID than the PDelay request (191/150)
EXCEPTION: GPTP [20:42:38:495] PDelay Response Receipt Timeout
ERROR    : GPTP [20:42:38:509] Source port identity from PDelay Response/FUP differ
ERROR    : GPTP [20:42:38:509] >>> Received PDelay followup but no RESPONSE exists
ERROR    : GPTP [20:42:38:704] Received PDelay FUP has different seqID than the PDelay request (228/151)
ERROR    : GPTP [20:42:39:220] Received PDelay FUP has different seqID than the PDelay request (192/151)
EXCEPTION: GPTP [20:42:39:504] PDelay Response Receipt Timeout
STATUS   : GPTP [20:42:39:511] AsCapable: Enabled
ERROR    : GPTP [20:42:39:511] >>> Received PDelay followup but no RESPONSE exists
ERROR    : GPTP [20:42:39:709] Received PDelay FUP has different seqID than the PDelay request (229/152)
ERROR    : GPTP [20:42:40:230] Received PDelay FUP has different seqID than the PDelay request (193/152)
ERROR    : GPTP [20:42:40:523] Source port identity from PDelay Response/FUP differ
ERROR    : GPTP [20:42:40:523] >>> Received PDelay followup but no RESPONSE exists
ERROR    : GPTP [20:42:40:720] Received PDelay FUP has different seqID than the PDelay request (230/153)
ERROR    : GPTP [20:42:41:241] Received PDelay FUP has different seqID than the PDelay request (194/153)
ERROR    : GPTP [20:42:41:539] Source port identity from PDelay Response/FUP differ
ERROR    : GPTP [20:42:41:539] >>> Received PDelay followup but no RESPONSE exists
ERROR    : GPTP [20:42:41:723] Received PDelay FUP has different seqID than the PDelay request (231/154)
ERROR    : GPTP [20:42:42:252] Received PDelay FUP has different seqID than the PDelay request (195/154)
STATUS   : GPTP [20:42:42:268] Switching to Slave
STATUS   : GPTP [20:42:42:268] New Grandmaster "68:05:CA:FF:FE:8D:32:98" (previous "68:05:CA:FF:FE:92:F6:FD")
ERROR    : GPTP [20:42:42:546] >>> Received PDelay followup but no RESPONSE exists
ERROR    : GPTP [20:42:42:726] Received PDelay FUP has different seqID than the PDelay request (232/155)**

I checked with the wire-shark and found that there is no unexpected Pdelay Request interfering in between a Pdelay Request and Pdelay Response Follow up message. In that case, why does my Sequence id not matching? and what should I do to solve this error?
   When I stopped one of the GPTP execution, then other two, are behaving as expected(get synchronized). Is there any modifications required in the gptp code, or am I missing anything in my setup to get synchronize more than 2 PCs.  Please have a look at the logs once and let me know if you could identify any mistake in it.
   Thanks in advance

PawelModrzejewski, 2019-10-04:
Does you switch support 802.1AS (gPTP)?

ram-nagapuri, 2019-10-04:
Hi Pawel, 
   Thank you for quick response. Currently I am using Netgear N300 wifi router(300mbps).  Is this matter?. It is working for 2 devices without any issues. or can I try any high speed(Gigabit) switch?

Thanks,
Ramesh

PawelModrzejewski, 2019-10-04:
You just can't use any switch, it must support at least IEEE 1588 / 802.1AS functionality. N300 is not that kind of device. You can play a little and try to set two devices as slaves and only one as master, but even if that works, you can face lots of other issues ...

ram-nagapuri, 2019-10-04:
We did tried 1 Master, 2 Slaves configuration, we are seeing issues with this also.
Do you have any preferred(low cost) 802.1AS complaint switches.

PawelModrzejewski, 2019-10-04:
Sorry, I'm not aware of any low cost switches but I've never searched for such devices ...

andrew-elder, 2019-10-04:
I think Motu and PreSonus both have low port count AVB switches.

ram-nagapuri, 2019-10-10:
Hi Pawel and andrew,
   Thank you for the quick response. Can you please help me to understand the use of AVB switches. How they are useful in running the GPTP on more than 2 devices.  I think, in WiFi (with AVB enabled), these switches will not be used. So, unsure of the purpose of these switches. Can you please point me to some documentation about this. 
Is there any alternative for this like, without using the (AVB) switch can we use more than 2 PCs for AVB?
   
Thanks,
Ramesh

andrew-elder, 2019-10-10:
@ram-nagapuri - I'm not an expert in AVB over wireless. Perhaps someone else reading this can comment.

In the wired case the switch runs 802.1AS (pPTP) on each port so that time is reproduced very accurately between devices. The timing across each network cable is measured independently and very accurately. The switch will also reserve bandwidth for AVB traffic and forward that traffic in a hardware queue that is independent from general purpose traffic. The network can be "swamped" and the AVB traffic will continue to flow without interruption.

A standard off-the-shelf switch will not run gPTP and will not support bandwidth reservation.