Avnu/OpenAvnu #890: Announcing the TSN Documentation Project for Linux

aguedes, 2019-11-20:
Hi folks,

For the past few years, several TSN features have been enabled in the Linux upstream ecosystem. To name some, Linux kernel mainline [1] provides hardware-agnostic interfaces to configure Credit-Based Shaper (CBS, former Qav), Enhancements for Scheduled Traffic (EST, former Qbv), and LaunchTime, Linux PTP [2] supports gPTP and Automotive profiles, ALSA [3] provides the AAF Plugin which enables applications to implement AVB audio use-cases, and GStreamer [4] provides the AVTP Plugin which enables applications to implement both AVB audio and video use-cases.

To help developers and system integrators to get started, we’ve put together a documentation project which provides hands-on, step-by-step guides on how to leverage what is available in the Linux upstream ecosystem. This documentation project stitches together the information scattered throughout cover letters and project-specific documentation in one single location.

Temporarily, the project is hosted in my GitHub account https://github.com/aguedes/tsn-doc and the documentation is available in https://tsn-doc.readthedocs.io/en/latest/.

We believe this project is interesting to the Avnu GitHub community so we'd like to contribute it. Hence, we’d like to inquire about your interest in hosting it under https://github.com/AVnu, just like we do with Libavtp project (https://github.com/AVnu/libavtp). If so, I’d like to ask the maintainers to create a new repository named "tsn-doc" under https://github.com/AVnu, to where I could move the project.

Best regards,

Andre

[1] https://www.kernel.org
[2] http://linuxptp.sourceforge.net
[3] https://www.alsa-project.org
[4] https://gstreamer.freedesktop.org

andrew-elder, 2019-11-20:
Hi Andre,
Thanks for doing this and I am 100% supportive. The repo has been created and you have write privileges.
Thanks again,
Andrew

aguedes, 2019-11-21:
Hi Andrew, thanks for your support. I moved the contents to the repo and set Read The Docs up to build the documentation and make it available in https://tsn.readthedocs.io. I couldn’t setup automatic builds, though, due to some missing configuration on the GitHub project. I need admin permissions to fix it. Could you please grant me that?

andrew-elder, 2019-11-21:
You should be good to go with admin privileges for the tsn-doc repo. Please let me know if it didn't work, or if you need anything else.
- Andrew

aguedes, 2019-11-21:
Great, I was able to fix it. Auto builds are now working. Thanks!

andrew-elder, 2019-11-22:
@aguedes - I'm going to assume you are happy to manage/merge PR requests to tsn-doc.

andrew-elder, 2019-11-22:
Can this issue be closed now?

aguedes, 2019-11-22:
Yes, we can. I'm closing it now. Thank you.

closed 2019-11-22 by aguedes