Avnu/OpenAvnu #892: software only AVB stack

vorio, 2020-01-14:
Are there any avnu certified software only AVB stack?

andrew-elder, 2020-01-15:
You can't really certify a software stack. The certification is attached to a final product. The nearest thing I am aware of that is no completely final product is the XMOS certification of their AVB evaluation module and the Coveloz certification of there FPGA based evaluation board.

vorio, 2020-01-15:
Just to clarify: There is already a L2/L3 network switch and need to add
this AVB feature. Is it possible to write software only solution given
strict latency needs? Or does it require FPGA solution on this existing
switch?
Similar to how Cisco supports:
https://www.cisco.com/c/en/us/products/switches/avb.html#~stickynav=1
They seem to have done AVB on FPGA.

pinealservo, 2020-01-15:
It's especially unlikely that you'll be able to implement AVB on a switch that wasn't designed with it in mind. In addition to the timestamping required, there are particular traffic shaping policies required, and these are usually built into the switch fabric hardware itself.  It's possible that your switch chip vendor has already implemented these in your switch fabric chip; most of them probably have the IP for it by now, even if it's not included in all their designs. In that case they may also have reference designs for the protocols involved as well, or may be able to refer you to a third party supplier. 

Software or even FPGAs can only help if they have the right kind of access to the right places in the overall hardware architecture, and nothing about switch hardware is standardized enough for us to give you a general answer here.

abin444, 2025-05-14:
Hi Vorio, it is unclear how the switch implementation is on your 'L2/L3 switch'. If it uses an off the shelf ethernet switch, first thing you need to know is whether the switch IC supports Hardware timestamping, support for IEEE 802.1AS
IEEE 802.1BA, IEEE 802.1Qav, IEEE 802.1Qat, IEEE 802.1Q, 802.1p etc. Then comes whether there is an application processor that is capable of running the AVB stack and AVTP ( since you expect it to be equivalent to AVNU certified bridge, ie, to be Milan interoperable). A generic Ethernet switch IC most likely will not support these features.  Experts, please correct me if I'm wrong.