Avnu/OpenAvnu #893: Organization name change

andrew-elder, 2020-02-06:
I have had a request to change the top level name of this repo from "AVnu" tp "Avnu". ie, correct the capitalization. The name change will probably break quite a few things. Any comments?

aguedes, 2020-02-06:
According to https://github.community/t5/How-to-use-Git-and-GitHub/Changing-username-case/td-p/6714, it seems the username is not case sensitive so it might not break things. But I'd recommend to verify this first. Break things is not good :)

christopher-s-hall, 2020-02-06:
The URL seems case insensitive

git clone https://github.com/avNU/OpenAvnu.git
git clone https://github.com/Avnu/OpenAvnu.git

Both work. Not what other considerations there might be.

andrew-elder, 2020-02-06:
Thanks for testing @christopher-s-hall - that was a good idea! I'll make the change then.

andrew-elder, 2020-02-06:
Name has changed. I'll keep this issue open in case there are some issues that I am unaware of at this time.

andrew-elder, 2020-02-26:
It's been almost 3 weeks. No side effects reported here at least. Closing this issue.

closed 2020-02-26 by andrew-elder