Avnu/OpenAvnu #901: build issue->no files in /opt/OpenAvnu-master/lib/igb_avb/lib

leonliu1128, 2020-08-31:
When I excute ./travis.sh to build, error print： 
make lib
make[1]: Entering directory '/opt/OpenAvnu-master'
mkdir -p lib/igb_avb/lib && make  subdir=lib/igb_avb/lib  -C lib/igb_avb/lib 
make[2]: Entering directory '/opt/OpenAvnu-master/lib/igb_avb/lib'
make[2]: *** No targets specified and no makefile found.  Stop.

Can anyone tell me the correct build step? Is travis.sh ok to run?

Thanks

mohankarthik, 2020-09-01:
Hi @leonliu1128 , this should mostly be because git submodules are not initialized. Could you try to initialize submodules and try again?

leonliu1128, 2020-09-07:
Hello mohankarthik:
Tks for feedback.
Can you let me know hot to initialize git submodules? I am new to openvnu.

leonliu2811, 2021-08-25:
Close this issue. I manually download code to the folder.