Avnu/OpenAvnu #902: OpenAvnu build on Nvidia Jetson Tx2

ajayinnovative2, 2020-10-30:
I am trying to build OpenAvnu for Nvidia Jetson Tx2 device which is aarch64 architecture having eth1000T1 (Marvel) network interface having driver lan743x (Marvel).  

During the build using travis.sh, it's failing for building igb_avb module with below error:
igb.c:1847:2: error: impossible constraint in 'asm'
__asm__ __volatile__("lfence;" 
Makefile:17: recipe for target 'gib.o' failed

My question is can OpenAvnu be build for non intel specific platform?
Is there any help doc available for building it for Nvidia platform and make use of various modules available in OpenAvnu?

ajayinnovative2, 2020-10-30:
Do I need to need change the igb dependencies in order to use avtp_pipeline?

andrew-elder, 2020-10-30:
I think you are kind of "on your own" if you are trying to run on a non-Intel platform. That said, I know someone got it working on an iMX6 or 7 platform sometime in the past. igb is short for "Intel something", so I'm not surprised that did not work out of the box.

I'm not aware of Nvidia specific docs at this time.

ajayinnovative2, 2020-10-30:
Do we have a way to disable igb dependencies and make the avtp pipeline work for non intel platforms.

andrew-elder, 2020-11-05:
I don't know how to do that.