Avnu/OpenAvnu #912: How to use the control points specified in "802.1AS Recovered Clock Quality Testing" ?

MNaveen9229, 2021-05-19:
Hi All,

For the recovered clock testing using "reverse synch" method , it  is specified in the "802.1AS Recovered Clock Quality Testing"  document to use different control points as follows,

1) A control point to enable and disable the Reverse Sync/Follow-up messages <reverseSyncEnabled>. The default value of this is FALSE. 
2) A control point to set the domain of Reverse Sync/Follow-up messages <reverseSyncDomain>. For certification it is proposed that a different unique domain should be used than the one that’s being  measured. When using in deployed networks a different domain shall be used. 
3) A control point to set the Reverse Sync/Follow-up message rate as % of incoming rate <reverseSyncRate>. The default is defined in Section 6.3.

Here does "reverseSyncEnabled","reverseSyncDomain" and "reverseSyncRate" refers to the control variables which has to be newly created , and based on this variables the code has to be updated ? 
(or) 
Does these control points are already available in the opensource code. I have searched for this variables in the code, but was unable to find them.

Any suggestions/ references to how to proceed on this will be really helpful.

PawelModrzejewski, 2021-05-19:
AFAIK AVnu/gptp doesn't support "reverse sync" method.
BTW: 802.1AS/gPTP code has been moved to separate/dedicated repository AVnu/gptp

MNaveen9229, 2021-05-19:
> 
> 
> AFAIK AVnu/gptp doesn't support "reverse sync" method.
> BTW: 802.1AS/gPTP code has been moved to separate/dedicated repository AVnu/gptp

Hi PawelModrzejewski,
Thanks for the comment. I will check the separate repository.
so for Recovered clock certification testing, is there any other method to be followed other than 1PPS?

Thanks in advance.