Avnu/OpenAvnu #928: Cannot build daemon_cl on aPTP branch on Raspberry Pi

adam-woolhether, 2023-06-14:
Using the `ArtAndLogic-aPTP-changes` branch.
I've run the initial steps:
```
git submodule init
git submodule update
mkdir tmp
cd tmp
cmake .. -G "Unix Makefiles"
make
```
Also, per `READMEaPTP.rst`:
```
ARCH=RPI make clean all
```

This doesn't build the desired gptp daemon.

I tried:
```
cd /daemons/gptp/linux/build
ARCH=RPI make clean all
``
Errors follow, here's build output:
```
root@raspberrypi:~/OpenAvnu/daemons/gptp/linux/build# ARCH=RPI make clean all
rm -f *~ obj/*.o  obj/daemon_cl
g++ -Wall -g -I. -I../../common -I../src -I/root/header/include/ -DPTP_HW_CROSSTSTAMP -DRPI -DAPTP -std=c++0x -Wnon-virtual-dtor  -c ../../common/ptp_message.cpp -o obj/ptp_message.o
../../common/ptp_message.cpp: In function 'std::shared_ptr<PTPMessageCommon> buildPTPMessage(char*, size_t, LinkLayerAddress*, EtherPort*, const Timestamp&)':
../../common/ptp_message.cpp:600:31: warning: 'void* memcpy(void*, const void*, size_t)' writing to an object of type 'class FollowUpTLV' with no trivial copy-assignment; use copy-assignment or copy-initialization instead [-Wclass-memaccess]
  600 |     sizeof(followup_msg->tlv) );
      |                               ^
In file included from ../../common/common_port.hpp:49,
                 from ../../common/avbts_clock.hpp:39,
                 from ../../common/ptp_message.cpp:35:
../../common/avbts_message.hpp:844:7: note: 'class FollowUpTLV' declared here
  844 | class FollowUpTLV {
      |       ^~~~~~~~~~~
g++ -Wall -g -I. -I../../common -I../src -I/root/header/include/ -DPTP_HW_CROSSTSTAMP -DRPI -DAPTP -std=c++0x -Wnon-virtual-dtor  -c ../../common/ap_message.cpp -o obj/ap_message.o
g++ -Wall -g -I. -I../../common -I../src -I/root/header/include/ -DPTP_HW_CROSSTSTAMP -DRPI -DAPTP -std=c++0x -Wnon-virtual-dtor  -c ../../common/avbts_osnet.cpp -o obj/avbts_osnet.o
g++ -Wall -g -I. -I../../common -I../src -I/root/header/include/ -DPTP_HW_CROSSTSTAMP -DRPI -DAPTP -std=c++0x -Wnon-virtual-dtor  -c ../../common/ether_port.cpp -o obj/ether_port.o
g++ -Wall -g -I. -I../../common -I../src -I/root/header/include/ -DPTP_HW_CROSSTSTAMP -DRPI -DAPTP -std=c++0x -Wnon-virtual-dtor  -c ../../common/common_port.cpp -o obj/common_port.o
g++ -Wall -g -I. -I../../common -I../src -I/root/header/include/ -DPTP_HW_CROSSTSTAMP -DRPI -DAPTP -std=c++0x -Wnon-virtual-dtor  -c ../../common/common_tstamper.cpp -o obj/common_tstamper.o
g++ -Wall -g -I. -I../../common -I../src -I/root/header/include/ -DPTP_HW_CROSSTSTAMP -DRPI -DAPTP -std=c++0x -Wnon-virtual-dtor  -c ../../common/ieee1588clock.cpp -o obj/ieee1588clock.o
../../common/ieee1588clock.cpp: In constructor 'IEEE1588Clock::IEEE1588Clock(bool, bool, uint8_t, uint8_t, uint8_t, OSTimerQueueFactory*, OS_IPC*, OSLockFactory*)':
../../common/ieee1588clock.cpp:130:64: warning: 'void* memset(void*, int, size_t)' writing to an object of type 'class ClockIdentity' with no trivial copy-assignment [-Wclass-memaccess]
  130 |   memset( &LastEBestIdentity, 0xFF, sizeof( LastEBestIdentity ));
      |                                                                ^
In file included from ../../common/ieee1588clock.cpp:34:
../../common/ieee1588.hpp:169:7: note: 'class ClockIdentity' declared here
  169 | class ClockIdentity {
      |       ^~~~~~~~~~~~~
../../common/ieee1588clock.cpp: In member function 'bool IEEE1588Clock::restoreSerializedState(void*, off_t*)':
../../common/ieee1588clock.cpp:227:64: warning: 'void* memcpy(void*, const void*, size_t)' writing to an object of type 'class ClockIdentity' with no trivial copy-assignment; use copy-assignment or copy-initialization instead [-Wclass-memaccess]
  227 |    memcpy( &LastEBestIdentity, buf, sizeof( LastEBestIdentity ));
      |                                                                ^
In file included from ../../common/ieee1588clock.cpp:34:
../../common/ieee1588.hpp:169:7: note: 'class ClockIdentity' declared here
  169 | class ClockIdentity {
      |       ^~~~~~~~~~~~~
g++ -Wall -g -I. -I../../common -I../src -I/root/header/include/ -DPTP_HW_CROSSTSTAMP -DRPI -DAPTP -std=c++0x -Wnon-virtual-dtor  -c ../src/linux_hal_common.cpp -o obj/linux_hal_common.o
../src/linux_hal_common.cpp: In member function 'bool LinuxLock::initialize(OSLockType)':
../src/linux_hal_common.cpp:772:10: warning: enum constant in boolean context [-Wint-in-bool-context]
  772 |   return oslock_fail;
      |          ^~~~~~~~~~~
g++ -Wall -g -I. -I../../common -I../src -I/root/header/include/ -DPTP_HW_CROSSTSTAMP -DRPI -DAPTP -std=c++0x -Wnon-virtual-dtor  -c ../src/linux_hal_persist_file.cpp -o obj/linux_hal_persist_file.o
g++ -Wall -g -I. -I../../common -I../src -I/root/header/include/ -DPTP_HW_CROSSTSTAMP -DRPI -DAPTP -std=c++0x -Wnon-virtual-dtor  -c ../../common/gptp_log.cpp -o obj/gptp_log.o
g++ -Wall -g -I. -I../../common -I../src -I/root/header/include/ -DPTP_HW_CROSSTSTAMP -DRPI -DAPTP -std=c++0x -Wnon-virtual-dtor  -c ../src/platform.cpp -o obj/platform.o
cc -Wall -g -I. -I../../common -I../src -I/root/header/include/ -DPTP_HW_CROSSTSTAMP -DRPI -DAPTP -c ../../common/ini.c -o obj/ini.o
g++ -Wall -g -I. -I../../common -I../src -I/root/header/include/ -DPTP_HW_CROSSTSTAMP -DRPI -DAPTP -std=c++0x -Wnon-virtual-dtor  -c ../../common/gptp_cfg.cpp -o obj/gptp_cfg.o
g++ -Wall -g -I. -I../../common -I../src -I/root/header/include/ -DPTP_HW_CROSSTSTAMP -DRPI -DAPTP -std=c++0x -Wnon-virtual-dtor  -c ../src/linux_hal_generic.cpp -o obj/linux_hal_generic.o
[... 93 lines cut ...]
                 from ../src/linux_hal_generic.cpp:37:
/usr/include/aarch64-linux-gnu/bits/socket.h: In member function ‘virtual net_result LinuxNetworkInterface::nrecv(LinkLayerAddress*, uint8_t*, size_t&)’:
/usr/include/aarch64-linux-gnu/bits/socket.h:285:33: error: flexible array member ‘cmsghdr::__cmsg_data’ not at end of ‘struct LinuxNetworkInterface::nrecv(LinkLayerAddress*, uint8_t*, size_t&)::<unnamed>’
  285 |     __extension__ unsigned char __cmsg_data __flexarr; /* Ancillary data.  */
      |                                 ^~~~~~~~~~~
../src/linux_hal_generic.cpp:63:8: note: next member ‘char LinuxNetworkInterface::nrecv(LinkLayerAddress*, uint8_t*, size_t&)::<unnamed struct>::control [256]’ declared here
   63 |   char control[256];
      |        ^~~~~~~
../src/linux_hal_generic.cpp:61:9: note: in the definition of ‘struct LinuxNetworkInterface::nrecv(LinkLayerAddress*, uint8_t*, size_t&)::<unnamed>’
   61 |  struct {
      |         ^
In file included from /usr/include/aarch64-linux-gnu/sys/socket.h:33,
                 from ../src/linux_hal_generic.cpp:37:
/usr/include/aarch64-linux-gnu/bits/socket.h: In member function ‘virtual int LinuxTimestamperGeneric::HWTimestamper_txtimestamp(PortIdentity*, PTPMessageId, Timestamp&, unsigned int&, bool)’:
/usr/include/aarch64-linux-gnu/bits/socket.h:285:33: error: flexible array member ‘cmsghdr::__cmsg_data’ not at end of ‘struct LinuxTimestamperGeneric::HWTimestamper_txtimestamp(PortIdentity*, PTPMessageId, Timestamp&, unsigned int&, bool)::<unnamed>’
  285 |     __extension__ unsigned char __cmsg_data __flexarr; /* Ancillary data.  */
      |                                 ^~~~~~~~~~~
../src/linux_hal_generic.cpp:290:8: note: next member ‘char LinuxTimestamperGeneric::HWTimestamper_txtimestamp(PortIdentity*, PTPMessageId, Timestamp&, unsigned int&, bool)::<unnamed struct>::control [256]’ declared here
  290 |   char control[256];
      |        ^~~~~~~
../src/linux_hal_generic.cpp:288:9: note: in the definition of ‘struct LinuxTimestamperGeneric::HWTimestamper_txtimestamp(PortIdentity*, PTPMessageId, Timestamp&, unsigned int&, bool)::<unnamed>’
  288 |  struct {
      |         ^
make: *** [Makefile:150: obj/linux_hal_generic.o] Error 1
```

MarcinMiklas, 2023-06-14:
We don't have here ArtAndLogic-aPTP-changes branch and RPI is not supported officially. 

The error you see is already fixed almost 3 years ago: 
https://github.com/Avnu/gptp/pull/32/files

I hope this helps.

adam-woolhether, 2023-06-14:
My mistake, I was referencing the branch from merge request in `feature-aptp-experimental`, sorry about the mixup.

I've now checked out the `feature-aptp-experimental` branch on  Avnu/OpenAvnu, and when attempting to build get:

```
g++ -Wall -g -I. -I../../common -I../src -I/root/header/include/ -DPTP_HW_CROSSTSTAMP -std=c++0x -Wnon-virtual-dtor  -c ../../common/ap_message.cpp -o obj/ap_message.o
g++ -Wall -g -I. -I../../common -I../src -I/root/header/include/ -DPTP_HW_CROSSTSTAMP -std=c++0x -Wnon-virtual-dtor  -c ../../common/avbts_osnet.cpp -o obj/avbts_osnet.o
g++ -Wall -g -I. -I../../common -I../src -I/root/header/include/ -DPTP_HW_CROSSTSTAMP -std=c++0x -Wnon-virtual-dtor  -c ../../common/ether_port.cpp -o obj/ether_port.o
g++ -Wall -g -I. -I../../common -I../src -I/root/header/include/ -DPTP_HW_CROSSTSTAMP -std=c++0x -Wnon-virtual-dtor  -c ../../common/common_port.cpp -o obj/common_port.o
../../common/common_port.cpp: In constructor 'CommonPort::CommonPort(PortInit_t*)':
../../common/common_port.cpp:57:32: error: 'const class OSThreadFactory' has no member named 'create'
   57 |  link_thread = thread_factory->create();
      |                                ^~~~~~
../../common/common_port.cpp:59:37: error: 'const class OSThreadFactory' has no member named 'create'
   59 |  listening_thread = thread_factory->create();
      |                                     ^~~~~~
make: *** [Makefile:192: obj/common_port.o] Error 1
```

MarcinMiklas, 2023-06-14:
As you probably already noticed OpenAvnu is not very actively developed. master branch still compiles and works fine, but experimental branches are, well, experimental so they might not compile anymore :(