Avnu/OpenAvnu #935: Raspberry pi 5 AVB

ValentinPintat, 2025-05-08:
Hello everyone,
I saw an old  [issue](https://github.com/AVnu/OpenAvnu/issues/656) saying that the raspberry pi 4 cannot implement AVB endpoint because it didn't have PCI Express or  didn't support 802.1AS.
I saw that the Raspi 5 does have a PCI express.
Does that means that it would be now possible to build an AVB endpoint on Raspi ?
The documentation also says : "10/100/1000 Mbps Ethernet MAC compatible with the IEEE 802.3 standard." does that means that the ethernet port can now be used to do AVB ?
Have a nice day

jdkoftinoff, 2025-05-08:
I purchased a Raspberry Pi 5 from adafruit.com and with the standard OS which came with it I was able to run linuxptpv2 with some configuration tweaks to sync to my MOTU AVB switch which in turn was synced to my TimeMachines GPS receiver as GM.   I did run into an issue however; while the ethernet MAC on this system has hardware timestamping, it always timestamps every incoming IEEE 802.1AS message regardless of the kind of message it is.  So I do encounter problems when ever the peer sends a Sync, then Announce, and then Sync_Followup - The announce message in between interferes with the timestamp of the sync.  If you are able to shut of BMCA on your peer (like in the automotive profile) then you will get more success.  If you just use the PCie port to connect a different MAC/PHY then you can choose an ethernet port which has more capabilities.

abin444, 2025-05-14:
A raspberry pi 5/CM5 connected through an Intel I210 over PCIe should be able to implement AVB, right? Anybody who thinks/experienced otherwise please comment.

jdkoftinoff, 2025-05-14:
Yes!